makermadecnc/makerverse #23: Minimum Z axis value default is -12.7 cannot change to anything less than 0

Orob-Maslow, 2020-09-01:
If $92 is the z axis depth plunge limit.  Defaulted at 12.7 mm, it is too shallow to cut 3/4 wood. 
 it should be limited to 20 or 25 mm. In attempting to change it to -25 mm or anything less than 0, it does the following

> $92=-20
error:4 (Value < 0)

When changed to a positive value, it accepts it and then it can no longer plunge below zero and any attempt to cut into the work piece is limited by the soft alarm.

Orob-Maslow, 2020-09-01:
The workaround is to disable the soft alarm 

$20=0

zaneclaes, 2020-09-01:
@Orob-Maslow you've got it backwards. `12.7mm` is the safe travel distance (above stock). The values are not allowed to be negative in GRBL. There is a second value, `$132=25.4` which represents the maximum plunge distance. If you're seeing soft alarms outside the -25.4 -> 12.7 bounds, something is wrong.

Orob-Maslow, 2020-09-01:
why was it -12.7 before I changed it?

zaneclaes, 2020-09-01:
... well, crap. Good catch. I had originally set the default value to negative because it made more sense. But then I discovered it couldn't be changed by Grbl due to the negative value. But I forgot to update the default value in the firmware 🤦 Anyways, the Due firmware has now been updated with the new default. If you don't feel like flashing again, just set `$92=12.7`

zaneclaes, 2020-09-01:
_re-open if the issue persists_

closed 2020-09-01 by zaneclaes

zaneclaes, 2020-09-01:
Update: actually, a new Due firmware flash is required. There was a related bug.