makermadecnc/makerverse #24: Makerverse Y axis "autochange"

Orob-Maslow, 2020-09-01:
When 2 similar systems are connected, there is some crosstalk/interference.  The system is homed and not moving, but the Y axis is toggling between 0 and -93.  Where did -93 come from?  If told to go home, the Y axis runs up for no reason other than Y is somehow changing to -93 with no user input.  When homed, the system sets -93 as the Y axis home value.  Once the second system serial port is disconnected, this 0 to -93 toggle stops, but it should never happen.

Orob-Maslow, 2020-09-01:
see video [here](https://forums.maslowcnc.com/t/makerverse-2-due-controllers-error-interference/14511)

zaneclaes, 2020-09-01:
I’m not sure about the “cross-talk” comment, but what I see in the video makes me think that your machine’s home position is not configured to 0,0. This is common on the Maslow Mega, because the home position (in that context) is the “extend chain lengths” position. Therefore, when you say “set home,” you are telling the Arduino that you are at 0, -95mm.

You can change this by using the new calibration, specifically the “define home” tab. You can set the home position to anywhere along the Y axis, but centered on the X axis. So once you use “Define Home,” then the settings for “default chain lengths” will change. And later on, when you press “Set Home,” you’re telling the machine you’re back at that position.

Orob-Maslow, 2020-09-01:
why does it switch between 0 and -93 when the other system sends data?

zaneclaes, 2020-09-01:
Hmm yeah, that part is interesting. Is the other tab actively connected to its PCB, which has a coordinate of 0,0,0? So the hypothesis here is that the other tab is somehow sending its position to the active workspace?

If you close the connection on the other tab, does the bug persist?

Orob-Maslow, 2020-09-01:
it only toggles when the other connection is open and it stops when it is disconnected, but the -93 offset persists.  Pressing home resets home to -93.  That number is totally random and yet it doesn't change.

zaneclaes, 2020-09-01:
> Pressing home resets home to -93.

Clarification needed: I've been trying to draw a distinction between **Define Home** (in the Calibration), and **Set Home** (the blue button). I'm not sure which one you're doing. You need to do the first one to fix the issue you're describing. That's what my last post was about.

Orob-Maslow, 2020-09-01:
I've done both multiple times and neither one makes a difference.  I'm not sure what else to do.  How do you clear the eeprom on the due?

zaneclaes, 2020-09-01:
Could you screenshot all your settings from the Maslow widget first? I'd like to dissect them.

Resetting EEPROM is same as Maslow, i.e., `$RST=*` for "all." In fact, I just made the button for it in the next release :)

Orob-Maslow, 2020-09-01:
This is probably an issue very few will come up against... having 2 Maslow Due's on the same controller.

I reset the first due with the rst command and the error message went away, then re homed and it looks good.  the offset toggle went away.

Orob-Maslow, 2020-09-01:
after resetting and then running both together, the Y coordinate on the right now goes to +93 on and off from zero.
M2 system settings (ttyACM0):
[image]
[image]
[image]

Due v1 system settings (with motors attached and operational on ttyACM1):

This system has a different z axis motor and the settings and speeds are different for it as is the resolution.
[image]
[image]

The system has a motor to motor spacing of 1809 mm with a 1219 mm x 1219 mm cut area with the beam spaced up at 301 mm

zaneclaes, 2020-09-02:
Okay, if the square cut area and short motor spacing was expected, then this looks right. Except for the negative Z value. Could you please upgrade to the latest firmware + Makerverse and try again? The current MaslowDue repo has version `20200905`. I fixed some bugs related to the Z axis, as well as some travel bugs.

I just did the following to test your situation:

- Installed firmware v20200905
- Installed Makerverse v1.1.1-a
- Set up the workspace, set width & height to 1219.2 (both)
- Opened calibration and used "Wipe Settings"
- Defined home using the given Motor Offset Y and dist between motors
- Observed that the X/Y position was correctly 0/0
- Moved around a bit. Re-homed. Etc.

Here are the settings, as Calibration applied them:
```
client> $81=1219.2
client> $82=1219.2
client> $91=3360
client> $46=97.9
client> $X
client> $94=2301
client> $84=1809.75
client> $83=301.63
feeder> $H
```

[image]

closed 2020-10-26 by zaneclaes