makermadecnc/makerverse #41: M2 CNC - Easel Generated GCode (simple square) Error

justinlevi, 2020-10-18:
I'm very new to working with a CNC so apologies if I'm missing something really obvious. I recently purchased an M2 CNC and have flashed the Arduino Due with the latest firmware from https://github.com/makermadecnc/MaslowDue

I have calibrated the machine as precise as I could. When generating a simple square GCode using Easel (https://easel.inventables.com) 

I quite often get the following error: 
"Click the resume button to resume program execution"
[image]

I don't see a "resume" button anywhere and many times the machine is completely unresponsive. I'm not sure if this is an issue with the makerverse software or with the M2. At a mimimim, might be a UI issue, at least with the error message and the instruction to click the "resume" button.

Thanks for any assistance.  

Here's what I see in the console:

```
(�:�4 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[GC:G0 G54 G17 G21 G�j� M9 T0 F0 S0]
ok
error:1 (Expected command letter)
[GC:G0 G54 G17 G21 G90 G94 Mj2� S0]
ok
�5A���220.998,184.997,-1.587|FS:127,0|Pn:X>
> G1 X-0.01999 Y12.00000 F5.0 (line=8)
error:1 (Expected command letter)
error:1 (Expected command letter)
```

Here is the gcode that was generated from Easel:

```
G20
G90
G1 Z0.15000 F9.0
G0 X11.98001 Y12.00000
G1 Z-0.06250 F9.0
G1 X11.79250 Y12.00000 F5.0
G1 X0.16751 Y12.00000 F5.0
G1 X-0.01999 Y12.00000 F5.0
G1 X-0.01999 Y11.81250 F5.0
G1 X-0.01999 Y0.18750 F5.0
G1 X-0.01999 Y0.00000 F5.0
G1 X0.16751 Y0.00000 F5.0
G1 X11.79250 Y0.00000 F5.0
G1 X11.98001 Y0.00000 F5.0
G1 X11.98001 Y0.18749 F5.0
G1 X11.98001 Y11.81251 F5.0
G1 X11.98001 Y12.00000 F5.0
G1 Z-0.12500 F9.0
G1 X11.79250 Y12.00000 F5.0
G1 X0.16751 Y12.00000 F5.0
G1 X-0.01999 Y12.00000 F5.0
G1 X-0.01999 Y11.81250 F5.0
G1 X-0.01999 Y0.18750 F5.0
G1 X-0.01999 Y0.00000 F5.0
G1 X0.16751 Y0.00000 F5.0
G1 X11.79250 Y0.00000 F5.0
G1 X11.98001 Y0.00000 F5.0
G1 X11.98001 Y0.18749 F5.0
G1 X11.98001 Y11.81251 F5.0
G1 X11.98001 Y12.00000 F5.0
G1 Z-0.17000 F9.0
G1 X11.79250 Y12.00000 F5.0
G1 Z-0.18750 F9.0
G1 X0.16751 Y12.00000 F5.0
G1 Z-0.17000 F9.0
G1 X-0.01999 Y12.00000 F5.0
G1 X-0.01999 Y11.81250 F5.0
G1 Z-0.18750 F9.0
G1 X-0.01999 Y0.18750 F5.0
G1 Z-0.17000 F9.0
G1 X-0.01999 Y0.00000 F5.0
G1 X0.16751 Y0.00000 F5.0
G1 Z-0.18750 F9.0
G1 X11.79250 Y0.00000 F5.0
G1 Z-0.17000 F9.0
G1 X11.98001 Y0.00000 F5.0
G1 X11.98001 Y0.18749 F5.0
G1 Z-0.18750 F9.0
G1 X11.98001 Y11.81251 F5.0
G1 Z-0.17000 F9.0
G1 X11.98001 Y12.00000 F5.0
G1 X11.79250 Y12.00000 F5.0
G1 Z-0.25000 F9.0
G1 X0.16751 Y12.00000 F5.0
G1 Z-0.17000 F9.0
G1 X-0.01999 Y12.00000 F5.0
G1 X-0.01999 Y11.81250 F5.0
G1 Z-0.25000 F9.0
G1 X-0.01999 Y0.18750 F5.0
G1 Z-0.17000 F9.0
G1 X-0.01999 Y0.00000 F5.0
G1 X0.16751 Y0.00000 F5.0
G1 Z-0.25000 F9.0
G1 X11.79250 Y0.00000 F5.0
G1 Z-0.17000 F9.0
G1 X11.98001 Y0.00000 F5.0
G1 X11.98001 Y0.18749 F5.0
G1 Z-0.25000 F9.0
G1 X11.98001 Y11.81251 F5.0
G1 Z-0.17000 F9.0
G1 X11.98001 Y12.00000 F5.0
G20
G90
G1 Z0.15000 F9.0
G0 X0.00000 Y0.00000
G4 P0.1
```

zaneclaes, 2020-10-26:
What version of Makerverse is this? It looks to be the old public release, from about 4 months ago, based upon the screenshot.

justinlevi, 2020-10-27:
Yes, this was the older public release. Any idea when the next public
release will be?

zaneclaes, 2020-10-27:
> Any idea when the next public release will be?

[v1.1.2](https://github.com/makermadecnc/makerverse/releases/tag/1.1.2) was released 3 days ago.

> have flashed the Arduino Due with the latest firmware from https://github.com/makermadecnc/MaslowDue

Note that you're presently using incompatible firmware + software.

I'd encourage you to read the docs; I just added about 3x more content yesterday: http://www.makerverse.com/

closed 2020-11-12 by zaneclaes