makermadecnc/makerverse #44: NO email confirmation received for Windows standalone installation

metalmaslow, 2020-10-26:
I downloaded the Version 1.1.1-g 1 (build #250) to a windows computer.
I created an account and it says to check conformation email. But I never get the email after 10 minutes
Tried another email address to make another account, same issue.

metalmaslow, 2020-10-26:
finally got email, just took longer than I thought.

closed 2020-10-26 by metalmaslow

zaneclaes, 2020-10-26:
In case anybody else runs across this —
I can pretty much guarantee that any delays are happening due to your email account, not due to Makerverse or OpenWorkShop. I created over 100 accounts during testing, and my gmail never took more than 30 seconds to receive an email. But my self-ran accounts (other email servers) sometimes took several minutes. This is normal; some receivers/forwarders are slow (I know to expect this with some of my email accounts; one in particular I rarely use for email confirmations on any site because it takes so long).

However, on the MV/OWS side, as soon as you see **check your inbox**, it means the software has already received confirmation that the email was sent. For example, if it's sending to `...@gmail.com`, then it means that `gmail.com` has confirmed that it received the email. My slow email is `@claes.biz`, and it is slow because the `claes.biz` server needs to forward the email along to the correct servers. 

I used to manage the team at a software company that sent the email notifications — millions of notifications per day — and we used the same tools (SendGrid) I used on OpenWorkShop. So it should hopefully be robust :)

neilferreri, 2020-11-12:
I'm not getting an email.  It's not my email.  I get gmail pretty much instantly from anywhere else.  GitHub sent me an email immediately about my linking the account.

zaneclaes, 2020-11-12:
@neilferreri I looked at the server logs. You did not include the `.com` when you typed your email address. The email address you provided ended in `@gmail` (instead of `@gmail.com`).

neilferreri, 2020-11-12:
Ha!  Well, at least it wasn't gmail's issue. I suggest a format check on the email field...I was actually testing it for you. ;)

zaneclaes, 2020-11-12:
Haha thanks ;) And indeed, you inspired me to implement SendGrid's email-validation tool, which lets us pre-check if the email is valid. It's much more advanced than a simple pattern checker and should hopefully prevent a much wider class of problems in the future...

neilferreri, 2020-11-12:
@zaneclaes So how do I procedd from here?  I linked my GitHub username, but when I try to login, I get:
[image]
Any chance you could force send the email?

zaneclaes, 2020-11-12:
@neilferreri use the "Forgot Password?" link on the login screen, and enter the same email address you used when you linked your GitHub account. It will re-send the verification link (rather than a forgot password link) since the account is not verified.

n.b., there is no way to "link" a Github account with an existing account. If you are trying to log in with Github, you must have also created the account with Github.

neilferreri, 2020-11-12:
@zaneclaes That won't work...The email I used was invalid due to my omitting the ".com"

zaneclaes, 2020-11-12:
> The email I used was invalid due to my omitting the ".com"

Typically, what I'd ask you to do here is just create a new account with the correct email. However, to make things easier, I just went into the database and manually edited the email address, as well as set the account as "validated." So you should be able to login with the (correct) email and password, now.

> I maintain the "CNCjs-Macros" repo on CNCjs.

Cool! Can't wait to hear your thoughts.