makermadecnc/makerverse #46: how to create user credentials with no networking

dlarue, 2020-10-31:
How do I setup my users using only local private LAN connections(no internet)?
Is there a tool to generate user credentials.

zaneclaes, 2020-10-31:
If the computer will ever be online, then you should login once while it is online and the login will continue to function forever once you go offline.

If the computer will **never** be online, then read the Security page from the docs. In there is explained how to enable `guest mode`, which disables login all together.

closed 2020-10-31 by zaneclaes

dlarue, 2020-10-31:
So whenever a user connects and wants to create an account the system has to be on the Internet but after each account creation it can be disconnected from the Internet?

zaneclaes, 2020-10-31:
> after each account creation it can be disconnected from the Internet?

Sounds roughly correct. Though, I might specify: _account login_ rather than _account creation_. As of this comment, Makerverse does not yet support multiple users accessing the same Makerverse installation. This will be coming soon.

There's one unlikely-but-possible scenario I'll explain, in case you're curious. If you're running the "Makerverse computer" as a web server and connecting to it from elsewhere in the house, each _account login_ (from a different computer/browser) will require that the _Makerverse computer_ be online. If the Makerverse computer goes offline for more than 30 days and then comes online again, the different computer/browsers may be forced to re-authenticate again. However, as long as the Makerverse computer is offline, it will treat an existing login (session) as valid, and the browser window will continue to work with Makerverse.

If you find any bugs or problems with this scenario, please feel free to open a new issue with the bug report or suggestion.