makermadecnc/makerverse #47: Add back CNCJS widget for manual feedrate adjustment please

simonodling, 2020-11-02:
Hi,
In Makerverse 1.0.6 there was a CNCJS widget in the left hand column that allowed feed rate (and spindle speed if used) to be adjusted by 1% or 10%, or reset to programmed value, during cutting. This was super helpful as when trying new cutters or if any issues presented during cutting you could turn down the feedrate, or up if desired during cutting.
It doesn't exist in 1.1.2, is there any possibility of turning it back on within 1.1.2 from the original CNCJS/Makerverse code?
I am running on Pi 4 using the latest 1.1.2 headless image.
Thanks in advance!
Simon

zaneclaes, 2020-11-06:
Hey @simonodling ! I added this to my todo list as soon as I saw it, but it seems I forgot to respond to you. Silly mistake on my part; I forgot to copy these "grbl" settings over into the "maslow" widget. Easy fix for the next release :)

simonodling, 2020-11-06:
Thanks @zaneclaes not sure we need the R field, most people will just use the F command, some may try and use spindle control, bit not many

neilferreri, 2020-11-12:
R is a Rapid rate override.  There are reasons to use it.