makermadecnc/makerverse #48: after creating an account I still can't login

dlarue, 2020-11-02:
I created an account with openwork.shop, got the email verification, clicked it and can't login at openwork.shop and can't log into my local instance of makerverse.

<clipped path>$ bin/makerverse 
2020-11-02T17:25:58.518Z - info init Loading configuration from "/home/dlarue/.makerverse"
2020-11-02T17:25:58.669Z - info init Starting the server at http://127.0.1.1:8000
2020-11-02T17:26:09.715Z - warn api:users Login Failure Result { data: {}, meta: { traceId: '0HM3P5AMDIQ41:00000024' } }
2020-11-02T17:26:09.716Z - warn api:users Login Error User not found
2020-11-02T17:26:09.992Z - warn api:users Login Failure Result { data: {}, meta: { traceId: '0HM3P5AMDIQ48:00000003' } }
2020-11-02T17:26:09.992Z - warn api:users Login Error User not found

zaneclaes, 2020-11-02:
Hi Doug,

Could you say more about "can't login at openwork.shop?" What exact error
are you receiving? I see that you pasted an error from Makerverse, but that
does not tell me what the error was when you tried to log in.

- Z

dlarue, 2020-11-02:
password or email are invalid

dlarue, 2020-11-02:
I also did the "forgot password" hussle , reset the password and that still didn't work.

zaneclaes, 2020-11-02:
> password or email are invalid

The error message should specify "username or password." Any chance you're trying to use your email (instead of username) to login?

dlarue, 2020-11-02:
the username was the email address.  I never got any errors registering and I got the confirmation email with the link to confirm. 
Again, no errors except when trying to login after clicking the confirmation link and all the effort after that.

dlarue, 2020-11-02:
none of the emails(confirmation nor reset password) show the userID.

closed 2020-11-02 by dlarue

zaneclaes, 2020-11-02:
> the username was the email address.

- You had to type manually the username yourself when you created the account.
- Usernames cannot contain a `@` character, so your email cannot be your username.

That said, I understand the confusion. I'm going to make a change so that you can use either the username or the email in order to login.

> none of the emails(confirmation nor reset password) show the userID.

That's a good point, thanks. I'm updating the code now to include a reminder of your username.

If you can't remember yours, shoot me an email (see my profile) and I can look it up for you.

wolfverines, 2021-11-26:
Bonjour,
cela fait des semaines que j'essaye de faire fonctionné maslow cnc M2, impossible de calibré correctement la machine 
j'ai essayé makerverse 1.0.6, 1.1.2, 1.1.3, 1.1.5 toujour des gros bugs de fonctionnement impossible de faire un simple carré de 200*200 (208*177 mm)
j'ai flasher le dernier firmware toujour pas de resultat
je voudrai repartir sur la version 1.1.2 et la .... impossible de me connecté "invalid password or credentials"
j'ai fait une demande de'un nouveau mot de passe toujour pas de connection 
j'ai crée 3 comptes different avec des adresses mail differentes toujour impossible de me connecté
la j'aurai besoin d'aide avant que je brule tout .....

Hello,
I have been trying to get maslow cnc M2 to work for weeks, unable to calibrate the machine correctly
I tried makerverse 1.0.6, 1.1.2, 1.1.3, 1.1.5 always big operating bugs impossible to make a simple square of 200 * 200 (208 * 177 mm)
I flashed the latest firmware still no result
I would like to start again on version 1.1.2 and the .... impossible to connect "invalid password or credentials"
I made a request for a new password still no connection
I created 3 different accounts with different email addresses still unable to connect
there I will need help before I burn everything .....

michael-barrow, 2022-05-22:
I had this issue. I opened an incognito window in chrome to access the server and the authentication worked. 

Probably something is wrong with auth tokens on a fresh install, I don't know. Something is for sure bogus because I am seeing this message from the docker image: warn api:users Login Error Internal Server Error in my docker log.