makermadecnc/makerverse #53: Unable to validate protocol (board not speaking at baud rate, or port is busy)

jokh0812, 2020-12-08:
I can't seem to get the Maslow Due to respond. I've started from scratch multiple times and I still keep getting the same result shown below. Any thoughts?
[image]

zaneclaes, 2020-12-09:
Have you verified that the firmware installed correctly, per the video/instructions? This "verification" step is doing the exact same thing shown in the video here: https://www.youtube.com/watch?v=WoopPgRBfx4&t=7s

jokh0812, 2020-12-09:
So I’ve done this a few times now and every time when I check the serial monitor it’s either blank or it’s nonsensical.

jokh0812, 2020-12-09:
To be honest, I’m not really that knowledgeable about coding. Am I suppose to adjust the baud rate on the firmware upload? 

Also, when I’ve done the upload before, as it finishes and the cpu resets the motors for the CNC start adjusting in tiny increments back and forth and it stops when I open the serial monitor to check that the firmware was uploaded properly. Is that an indicator of something or is that normal?

zaneclaes, 2020-12-09:
> when I check the serial monitor it’s either blank or it’s nonsensical.

This indicates the firmware has not been installed.

> Am I suppose to adjust the baud rate on the firmware upload?

No. Baud Rate is specified by the firmware itself.

> Also, when I’ve done the upload before, as it finishes and the cpu resets the motors for the CNC start adjusting in tiny increments back and forth and it stops

Hm, this could also happen even if the firmware install failed. You mentioned the CPU resetting, which sounds like a successful installation, but I still want to check that is the case.

In the video, you'll notice the "installation progress" in the window before I open the serial monitor. This is what tells me if the firmware installation was actually successful or not. I only open the Serial Monitor as a way to perform an additional validation, i.e., demonstrate what Makerverse is doing behind-the-scenes in order to communicate with the firmware.

Anywho, could you post your console output? Mine looks something like this: 

```
Linking everything together...
/Users/zaneclaes/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -Os -Wl,--gc-sections -T/Users/zaneclaes/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x/linker_scripts/gcc/flash.ld -Wl,-Map,/var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/MaslowDue.ino.map -o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/MaslowDue.ino.elf -L/var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--start-group -u _sbrk -u link -u _close -u _fstat -u _isatty -u _lseek -u _read -u _write -u _exit -u kill -u _getpid /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/DueTimer.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/MaslowDue.ino.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/coolant_control.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/eeprom.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/gcode.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/jog.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/limits.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/motion_control.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/nuts_bolts.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/planner.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/print.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/probe.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/protocol.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/report.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/serial.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/settings.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/sleep.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/spindle_control.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/stepper.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/sketch/system.cpp.o /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/core/variant.cpp.o /Users/zaneclaes/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x/libsam_sam3x8e_gcc_rel.a /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/../arduino_cache_607682/core/core_arduino_sam_arduino_due_x_dbg_4cb90ae67dc5acf8c5f9c0ae0a1d76e8.a -Wl,--end-group -lm -lgcc
/Users/zaneclaes/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-objcopy -O binary /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/MaslowDue.ino.elf /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/MaslowDue.ino.bin
/Users/zaneclaes/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-size -A /var/folders/05/_r73lmmd7_16nh9tyb43s5800000gn/T/arduino_build_640684/MaslowDue.ino.elf
Sketch uses 71316 bytes (13%) of program storage space. Maximum is 524288 bytes.
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.034 seconds

Write 73896 bytes to flash (289 pages)

[                              ] 0% (0/289 pages)
[==                            ] 9% (28/289 pages)
[=====                         ] 19% (56/289 pages)
[========                      ] 29% (84/289 pages)
[===========                   ] 38% (112/289 pages)
[==============                ] 48% (140/289 pages)
[=================             ] 58% (168/289 pages)
[====================          ] 67% (196/289 pages)
[=======================       ] 77% (224/289 pages)
[==========================    ] 87% (252/289 pages)
[============================= ] 96% (280/289 pages)
[==============================] 100% (289/289 pages)
done in 14.220 seconds

[... cut ...]
[===========                   ] 38% (112/289 pages)
[==============                ] 48% (140/289 pages)
[=================             ] 58% (168/289 pages)
[====================          ] 67% (196/289 pages)
[=======================       ] 77% (224/289 pages)
[==========================    ] 87% (252/289 pages)
[============================= ] 96% (280/289 pages)
[==============================] 100% (289/289 pages)
Verify successful
done in 13.023 seconds
Set boot flash true
CPU reset.
```

jokh0812, 2020-12-09:
This is all that is on the output when I run mine. And after watching the video more closely, I just realized that in the Tools menu under Programmer you have something listed. When I pull mine up, it says that there are no programmers found for this device.

Sketch uses 71316 bytes (13%) of program storage space. Maximum is 524288 bytes.
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.037 seconds

Write 73896 bytes to flash (289 pages)
[==============================] 100% (289/289 pages)
done in 14.220 seconds

Verify 73896 bytes of flash
[==============================] 100% (289/289 pages)
Verify successful
done in 14.129 seconds
Set boot flash true
CPU reset.

jokh0812, 2020-12-09:
Also, in the midst of my frustration I’m forgetting to say thank you for your help! I am at a loss and am glad to have somebody else to help troubleshoot.
So thank you for your timely replies!

zaneclaes, 2020-12-09:
> I just realized that in the Tools menu under Programmer you have something listed. When I pull mine up, it says that there are no programmers found for this device.

Hmm... here's a closer look at mine:
[image]

LMK if you're missing them or what. This seems like a good thing to pursue.

> So thank you for your timely replies!

Glad to help. Ironically, I'm working on a "install firmware" feature for Makerverse right now. It should (in theory) eliminate the need for the Arduino IDE all together, and make it a simple button-press.

jokh0812, 2020-12-09:
> 	LMK if you're missing them or what. This seems like a good thing to pursue.
> 
> Yea mine shows no programmers available. Not sure what to do with that

		Glad to help. Ironically, I'm working on a "install firmware" feature for Makerverse right now. It should (in theory) eliminate the need for the Arduino IDE all together, and make it a simple button-press.
      
That would be amazing! Over the past couple of years, my only major system crippling issues that I’ve run into have been Arduino IDE based. Haha

zaneclaes, 2020-12-09:
> Yea mine shows no programmers available. Not sure what to do with that

First place I'd start is the [Arduino Due Guide](https://www.arduino.cc/en/Guide/ArduinoDue) (from the [MaslowDue README](https://github.com/makermadecnc/MaslowDue)). There may be some step in there necessary for your platform.

Folks on StackExchange [seem to indicate](https://arduino.stackexchange.com/questions/78247/no-programmers-available-for-this-board-programming-esp32-cam-with-ardunio-ide) a recent Arduino IDE changed some behavior re: programmers. I'll upgrade my IDE and see if I run into the same issue.

jokh0812, 2020-12-09:
So I rolled back to IDE 1.8.12 and programmer options are back but nothing else has changed. It still does the same thing. Could there be a hardware issue that I’m unaware of?

zaneclaes, 2020-12-09:
Upon upgrading my Arduino IDE, I experienced the same problem (_no programmers available_). However, I was able to successfully flash my Arduino Due despite the lack of programmers.

At this point, my best guess is hardware, as well. What exact board are you using? Have you upgraded, or otherwise used this board elsewhere? Whatever info seems relevant.

jokh0812, 2020-12-09:
It’s the Arduino Due board that came with the MakerMade M2 upgrade. I have not used it anywhere else. I’ve had it in the box for over a month because I hadn’t had the time to make the transition. I started the upgrade process and have had this issue from the start

zaneclaes, 2020-12-09:
Hm. Just to be sure, I'd try using a different USB cable. I'd also make sure the shield is attached while you try to test.

It's strange to me that the IDE is able to "successfully" flash. It almost seems like something is preventing the Due from booting. Spitting out messages into the console should be one of the first things it does. However, I'd guess that various things could prevent the firmware from booting to that point (even something as simple as a shorted pin).

jokh0812, 2020-12-09:
Ok so I tried a different USB cable. And made sure the shield is attached. Outcome is the exact same. I don’t know enough about the hardware to really troubleshoot it. What would you suggest is the next step?

zaneclaes, 2020-12-09:
TBH, I'm more of a software guy than hardware myself. But my next debugging steps, in your shoes, would be to try to install some other Arduino sketch that runs on a Due board. I'm sure you can find some with some googling. You could even try installing the "normal" Grbl firmware, just to see if you can get the board to install _something_.

If you can't get that working, I'd ask the Makermade folks for an exchange.