makermadecnc/makerverse #54: Docker: toomanyrequests launching from the Raspberry Pi image

pwsh, 2020-12-13:
It looks like the project is hitting its rate limit from docker. This is a fresh install of a raspberry pi image using the 1.1.2 desktop img

Dec 13 15:21:23 makerverse runuser[9715]: pam_unix(runuser-l:session): session opened for user pi by (uid=0)
Dec 13 15:21:24 makerverse runuser[9715]: Makerverse:latest settings: /home/pi/.makerverse (project at: /home/pi/makerverse)
Dec 13 15:21:24 makerverse runuser[9715]: Docker v20.10.0 detected
Dec 13 15:21:24 makerverse runuser[9715]: Removing old versions (pruning docker)...
Dec 13 15:21:24 makerverse runuser[9715]: Total reclaimed space: 0B
Dec 13 15:21:24 makerverse runuser[9715]: Updating makerverse/core:latest...
Dec 13 15:21:25 makerverse dockerd[712]: time="2020-12-13T15:21:25.390289600-06:00" level=error msg="Not continuing with pull after error: toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating
Dec 13 15:21:25 makerverse dockerd[712]: time="2020-12-13T15:21:25.407605410-06:00" level=error msg="Handler for POST /v1.41/images/create returned error: toomanyrequests: You have reached your pull rate limit. You may increase the limit by
Dec 13 15:21:25 makerverse runuser[9715]: Error response from daemon: toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docker.com/increase-rate-limit
Dec 13 15:21:25 makerverse runuser[9715]: pam_unix(runuser-l:session): session closed for user pi
Dec 13 15:21:25 makerverse systemd[1]: makerverse.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit makerverse.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Dec 13 15:21:25 makerverse systemd[1]: makerverse.service: Failed with result 'exit-code'.

pwsh, 2020-12-13:
Manually rebuilding results in the same error

pi@makerverse:~ $ makerverse/bin/launch
Makerverse:latest settings: /home/pi/.makerverse (project at: /home/pi/makerverse)
Docker v20.10.0 detected
Removing old versions (pruning docker)...
Total reclaimed space: 0B
Updating makerverse/core:latest...
Error response from daemon: toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docker.com/increase-rate-limit

pwsh, 2020-12-13:
This was a new limit put in last month. They have a banner on their page for this also. [https://www.docker.com/increase-rate-limits](https://www.docker.com/increase-rate-limits)

zaneclaes, 2020-12-13:
Yikes, thanks for the heads up on that @pwsh. I just upgraded to the unlimited pulls Pro plan, so hopefully that should fix it.

closed 2020-12-13 by zaneclaes

pwsh, 2020-12-13:
Thanks! Looks like it might take some time to replicate in their system as I'm still receiving the rate limit message

Dec 13 17:14:14 makerverse runuser[23584]: pam_unix(runuser-l:session): session opened for user pi by (uid=0)
Dec 13 17:14:14 makerverse runuser[23584]: Makerverse:latest settings: /home/pi/.makerverse (project at: /home/pi/makerverse)
Dec 13 17:14:14 makerverse runuser[23584]: Docker v20.10.0 detected
Dec 13 17:14:14 makerverse runuser[23584]: Removing old versions (pruning docker)...
Dec 13 17:14:14 makerverse runuser[23584]: Total reclaimed space: 0B
Dec 13 17:14:14 makerverse runuser[23584]: Updating makerverse/core:latest...
Dec 13 17:14:15 makerverse dockerd[878]: time="2020-12-13T17:14:15.762061145-06:00" level=error msg="Not continuing with pull after error: toomanyrequests: You have reached your pull rate limit. You may i
Dec 13 17:14:15 makerverse dockerd[878]: time="2020-12-13T17:14:15.807550262-06:00" level=error msg="Handler for POST /v1.41/images/create returned error: toomanyrequests: You have reached your pull rate
Dec 13 17:14:15 makerverse runuser[23584]: Error response from daemon: toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docke
Dec 13 17:14:15 makerverse runuser[23584]: pam_unix(runuser-l:session): session closed for user pi
Dec 13 17:14:15 makerverse systemd[1]: makerverse.service: Main process exited, code=exited, status=1/FAILURE

zaneclaes, 2020-12-13:
Hm, let me know if it doesn't go away soon. I just did a `docker pull makerverse/core:latest` on a random machine (where I'm not logged in) and it worked, so maybe just caching...

pwsh, 2020-12-13:
I'll probably just try tomorrow. 

Tried purging cache, which there was none. 
Tried rebooting, no change. 
Tried a manual pull, no change. 
Tried on another machine, same error.

pi@makerverse:~ $ docker pull makerverse/core:latest
Error response from daemon: toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docker.com/increase-rate-limit

zaneclaes, 2020-12-13:
> Error response from daemon: toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docker.com/increase-rate-limit

Actually... are you sure it's not your account? According to their website:

> Anonymous and Free Docker Hub users are limited to 100 and 200 container image pull requests per six hours.

That sounds more like the limit is on the side of the puller, not the uploader...

pwsh, 2020-12-13:
Figured it out. The logic was attempting a pull every few seconds, so it looks like it temporarily banned my IP as it was just sitting in the background trying over and over for who knows how long. I connected it to a wifi network I have that routes all traffic through VPN and it worked. Docker allowed the pull after switching IPs.

I think it was a combination if issues here as I don't pull that much from docker on a daily basis. I think the rate limit was hit on both sides.

Dec 13 17:30:52 makerverse runuser[19021]: pam_unix(runuser-l:session): session opened for user pi by (uid=0)
Dec 13 17:30:53 makerverse runuser[19021]: Makerverse:latest settings: /home/pi/.makerverse (project at: /home/pi/makerverse)
Dec 13 17:30:53 makerverse runuser[19021]: Docker v20.10.0 detected
Dec 13 17:30:53 makerverse runuser[19021]: Removing old versions (pruning docker)...
Dec 13 17:30:53 makerverse runuser[19021]: Total reclaimed space: 0B
Dec 13 17:30:53 makerverse runuser[19021]: Updating makerverse/core:latest...
Dec 13 17:31:04 makerverse runuser[19021]: latest: Pulling from makerverse/core
Dec 13 17:31:05 makerverse runuser[19021]: 1b636fdf3723: Pulling fs layer
Dec 13 17:31:05 makerverse runuser[19021]: 9be3bc01c2e2: Pulling fs layer
...
Dec 13 17:31:09 makerverse runuser[19021]: 5e1b23b3e38f: Verifying Checksum
Dec 13 17:31:09 makerverse runuser[19021]: 5e1b23b3e38f: Download complete
Dec 13 17:31:11 makerverse runuser[19021]: 9be3bc01c2e2: Verifying Checksum
Dec 13 17:31:11 makerverse runuser[19021]: 9be3bc01c2e2: Download complete

zaneclaes, 2020-12-14:
> The logic was attempting a pull every few seconds

Ah, sounds like it was in a crash loop and restarting the service for some reason.

pwsh, 2020-12-14:
Working great now! Thanks again!