makermadecnc/makerverse #57: Macro Widget

neilferreri, 2021-01-11:
Why has the Macro Widget been removed?

zaneclaes, 2021-01-11:
There was no such thing when I inherited the code base. There were some lingering settings in the "Settings" section, but it looks like someone had already begun removing it from CNCjs.

Do you have any documentation about how it worked, or screenshots? I'd like to re-add it.

neilferreri, 2021-01-11:
https://github.com/cncjs/cncjs/tree/master/src/app/widgets/Macro
https://github.com/cncjs/cncjs/wiki/User-Guide#macro-widget
https://github.com/cncjs/CNCjs-Macros

makerorne, 2021-03-03:
it would be very usefull to have this feature because sometimes, depending on material, it is really easier to modulate the infeed speed "live" instead of leaving makerverse to make a new nc file.       thanks a lot for your work