makermadecnc/makerverse #59: Start router with gpio....

makerorne, 2021-02-18:
I want to wire relay on gpio of my raspberry pi with MAKERVERSE but i cant succeed in making it work...
i use a raspberry pi lite image of makerverse and i tried as described on manual but it dont work: i wired my relay on GPIO 3 and i try those commands ine the "setting \command menu:
first command is:
"start_router"
bash /home/node/makerverse/commands/gpio.sh 3 1
second is:
"stop router"
bash /home/node/makerverse/commands/gpio.sh 3 0
but i dont know how to trigger it for a try... there are no "command button" maybe i have to type command somewhere ont tne interface but i dont know where (it is not explained on the doc)
i found "event" feature which can start a command when something happens (start programm or other condition...) but it dont work for the moment.
maybe i have to wire my relay on another GPIOpin ? (wiring is 5v on v+, gnd to gnd and +signal on gpio 3 with a 5v relay card)
if someone has done that, i am very interested 😊
thank you
(on image "led " is the router)
[image]

zaneclaes, 2021-02-18:
>  (it is not explained on the doc)

It's the very first sentence of [the docs on Commands](http://www.makerverse.com/features/commands/#commands), which notes that they are triggered manually via "the dropdown in the top right." I apologize if that was not clear; here's a screenshot:

[image]

> i found "event" feature which can start a command when something happens (start programm or other condition...) but it dont work for the moment.

What do you mean it "dont work for the moment?" Just that you don't see the relay switch turn on/off as you would expect? In that case, the problem is likely in the wiring.

> maybe i have to wire my relay on another GPIOpin ? (wiring is 5v on v+, gnd to gnd and +signal on gpio 3 with a 5v relay card) if someone has done that, i am very interested 

In your diagram, GPIO 3 is not connected to anything. Did you intentionally use a diagram which differs from your actual wiring?

makerorne, 2021-02-19:
thanks for help !!! so with your screenshot it is more clear; i will try tonight and i will tell you
My diagram is an example from the web but i use really gpio 3 (pin5) and my wiring seems good (i use the same wiring on a jeedom domotic application on another pi)

and just one more thing (it is really difficult to find help about those function...): in "event" if i want to start router when program start, i add one event trigered by "program start" and i choose "system" and on the bottom of event i write the name a a previously made command (as "start router" in my case) or i retype the command ("bash /home/node/makerverse/commands/gpio.sh 3 1") ?
THANKS A LOT FOR YOUR HELP

a last thing; is it possible to try thoses commands without being connected to the M2 (i want to test this first on my desk if possible)

makerorne, 2021-02-19:
i tried it on gpio19 as the axample; it is better: i secced in switching it on but impossible to switch off... ???
th switch off i use: bash /home/node/makerverse/commands/gpio.sh 19 0

[image]

makerorne, 2021-02-19:
hey !! i understand: it was because i need to wire the VCC relay to 3.3V ! (with a 5V relay it work)
using 5v is not good: the gpio pin can just work with 3.3v , so now all is ok: commands are ok.
i will try with events and it will be a really good thing for my M2.
Thanks

zaneclaes, 2021-02-19:
Very glad you got it working! I'll post some thoughts here, in case it helps you or anyone else who stumbles on this...

Googling "Raspberry Pi Pinout" should give you a diagram like this. Despite building with Pis for years now, I still search for this weekly :P 

[image]

> i tried it on gpio19 as the axample; it is better: i secced in switching it on but impossible to switch off... ???

FYI, most (all?) relays have an LED indicator light. This will tell you if the relay is on/off so that you can know if the script is working without the relay actually connected to the machine.

> is it possible to try thoses commands without being connected to the M2

The easiest way to do this would be to open a terminal (command prompt) on the Raspberry Pi and simply run the same command/event script that you entered in Makerverse. The only caveat is that you should be in the Makerverse folder first, i.e., `cd ~/makerverse`.

> hey !! i understand: it was because i need to wire the VCC relay to 3.3V ! (with a 5V relay it work)

Ah, yep, it's important to check the specs! Note that the Raspberry Pi also has both 3.3V and 5V pinouts, so you could still use the 3.3V relay by attaching it to the correct RPi pin (see above).

closed 2021-02-19 by zaneclaes

makerorne, 2021-02-20:
Thanks a lot for your help !