makermadecnc/makerverse #61: User handling

3kmch, 2021-04-20:
I was finally able to set up the Makerverse on a Raspi and connect it to the Due. There are still some questions:

#0 I set up the Makerverse for a customer. Therefore I generated a user. Now I would like to add a new user. That seemed to work but we were not able to log in with it. Any hints?

#1 As soon as the new user works, can the old one get deleted?

#2 Is it possible to watch the milling process via mobile phone while beeing connected to a notebook?

zaneclaes, 2021-04-20:
1&2) At present, only one user account may be attached to your Makerverse account. There is not yet an official process for replacing one user with another. You would need to either edit the `~/.makerverse` file to remove the old user, or just delete it to clear the settings.

3) Yes. I do something similar (I use my phone on the shop floor, and then return to my desk to monitor the cut on my laptop). However, the current version doesn't look great on a phone. The upcoming version will be much more mobile optimized.

3kmch, 2021-04-21:
OK and thank you for your reply. :)

closed 2021-05-14 by zaneclaes