makermadecnc/makerverse #81: Getting error: Certificate is expired when trying to login to stadalone app

jdhoneatt, 2021-10-01:
Getting the error, "Certificate is expired" when trying to login to standalone app on Windows 10 computer.  Have tried installing latest version.  Was hoping to get some work done today...  Any guidance you can provide would be great!

[image]

zaneclaes, 2021-10-01:
tl;dr — the fix went live about 5 minutes ago. Please try again and LMK how it goes.

Full explanation: the problem had to do with Electron, and affected only the desktop (stand-alone) versions of Makerverse. Dozens of different desktop apps around the world are currently dead in the water, as you can read about here: https://github.com/electron/electron/issues/31212

zaneclaes, 2021-10-01:
We've had two other users verify that things are working again, so I'm going to close this issue for now. Feel free to ping me if needed @jdhoneatt .

closed 2021-10-01 by zaneclaes

jdhoneatt, 2021-10-01:
I can confirm it is working for me too.  Thanks for the quick response!