makermadecnc/makerverse #82: Network Error

Ddoliver135, 2021-10-10:
I just upgraded to M2, was using ground control with M1. I have installed Makerverse but when I press login/create account it just says network error. I have uninstalled and reinstalled but not luck. Any suggestions?

Ddoliver135, 2021-10-11:
I’m running OS X El Capitan v10.11.16

zaneclaes, 2021-10-11:
> For context, the only thing the button is doing is linking you to
> https://OpenWork.shop … the error you are describing comes from your OS
> (operating system) and indicating that it cannot contact the website. The
> error is not coming from Makerverse, but directly from your computer saying
> it cannot access that website. I have not heard of this happening before on
> MacOS. The only other possibilities I can think of would be some OS
> security setting, a network level security setting, or a browser security
> setting. I assume you are using the Desktop version of Makerverse, so this
> would be related to the Safari browser settings.
>
> But again, literally all the button is doing is linking you to a website —
> and the error is not coming from Makerverse, but from the network stack of
> your computer literally saying it cannot contact the website.

If you can’t figure it out (and can’t try a different computer), I’d
recommend following the instructions linked on the login screen to turn off
login-based security.