makermadecnc/makerverse #85: https://www.makerverse.com/ is serving the wrong HTTPS certificate

rspeed, 2021-11-06:
Specifically, it's serving a certificate for www.github.com. Compounding this issue is the fact that many links on the site have hard links to that domain (which is itself problematic). As a result, doing things such as attempting to read documentation often results in error messages about a failed TLS connection.

zaneclaes, 2021-11-17:
> Specifically, it's serving a certificate for www.github.com.

This is actually a requirement of websites hosted by Github Pages using https. We've had a lot of problems with Github Pages hosting, and plan to switch over to a permanent solution (our own host) with the next major release. Thanks for the report though!

closed 2021-11-17 by zaneclaes

rspeed, 2022-01-07:
Okay… but the links are still broken. At the very least they should have the domain removed so people can actually read the documentation.