makermadecnc/makerverse #86: Z axis can't be calibrated

rspeed, 2021-11-15:
I'm trying to set up a Maslow v1 with Makerverse, but the Z axis step in the calibration process is broken. Specifically, any time I try to move the Z axis the X and Y motors hum (but don't move) while there's no movement from the Z axis. I was able to skip this step and successfully calibrate the X/Y axes, but still no luck with Z. I tried upgrading to the current prerelaese version, but the issue remains.

zaneclaes, 2021-11-17:
This problem is not with Makerverse, but with the hardware/firmware. All Makerverse does is to issue a command to the machine (in this case, `G0 Z5` for example). The fact that the motors "hum" indicates that the command was received and processed by the board, but something is preventing it from being able to move. 

- Have you used this Mv1 before?
- What firmware version are you running?
- Did you make any changes to settings?

You can try resetting the board settings (EEPROM) from Makerverse and moving the Z axis directly.

closed 2022-01-04 by zaneclaes

rspeed, 2022-01-07:
This ended up being caused by the `z axis Kp Pos` ($29) setting somehow defaulting to 0. I don't know how it happened, as it had been working before setting up Makerverse, but I'm leaning towards it being something in the firmware.