makermadecnc/makerverse #89: USB webcam not passed through to Makerverse

shiftylilbastrd, 2021-12-10:
Hardware
    RPi 4 4gb
    ELP USB Camera 2 megapixel
Software
    Makerverse 1.1.3

Used makerverse-raspberry-pi-os-desktop-1.1.2.img.gz to flash SD card
Webcam is functioning via webtest and raspistill on Pi itself

Webcam not available via "built-in camera" option in Makerverse
Have established webserver via https://github.com/Motion-Project/motion as a test as well and can connect in Makerverse that way.

Still a Docker newbie but from what I've come to understand '-v /dev:/dev' in the launch file should be passing all devices through to the container but for some reason it's not.

skilescm, 2022-02-10:
Has this issue been resolved?

shiftylilbastrd, 2022-02-10:
Not that I’m aware of