makermadecnc/makerverse #90: Original Maslow Alarm: 3 (Abort during cycle)

buck2825, 2021-12-15:
My original maslow works fine under webcontrol.  In makerverse it calibrates and moves just fine.  I get a few lines into my GCode and I get an Alarm: 3 (Abort during cycle) displays in the console, and a Gray box with "Abort during cycle Reset while in motion. Machine position is likely lost due to sudden halt. Re-homing is highly recommended."

rspeed, 2022-01-01:
I'm having the same problem.

zaneclaes, 2022-01-04:
Does this always happen on the same line of the file? If so, it is almost certainly a problem with the GCode. Does it happen with other files? Have you tried this exact file with WebControl?

buck2825, 2022-01-04:
yes same line every time.  yes it happens with other files.  It runs well in webcontrol. I switched to webcontrol to finish cutting the project.

The line it tripped on was a simple G01.

zaneclaes, 2022-01-04:
Could you upload the file for me to test? 

One idea is to try copy & pasting the GCode (line by line) into the console. If it happens there, too, we know it's a problem with the Makerverse sending code (as opposed to the file reader). Note that the current version of Makerverse is a fork of CNCjs... and if you search Google for "Alarm: 3" the very first result is [an issue there](https://github.com/cncjs/cncjs/issues/137), which may be related.

buck2825, 2022-01-05:
`(Exported by FreeCAD)
(Post Processor: PathScripts.post.grbl_G81_post)
(Output Time:2021-12-16 14:56:31.177383)
(begin preamble)
G17 G90
G21
(begin operation: Job)
(Path: Job)
M1
G0 Z6.000
G0 X154.844 Y148.671 Z6.000
G0 X154.844 Y148.671 Z4.000
G1 X154.844 Y148.671 Z-1.117 F60.00
G1 X204.587 Y119.952 Z-1.117 F60.00
G2 X206.175 Y117.202 Z-1.117 I-1.587 J-2.750 K0.000 F60.00
G1 X206.175 Y-117.202 Z-1.117 F60.00
G2 X204.587 Y-119.952 Z-1.117 I-3.175 J0.000 K0.000 F60.00
G1 X1.587 Y-237.154 Z-1.117 F60.00
G2 X0.822 Y-237.471 Z-1.117 I-1.588 J2.750 K0.000 F60.00
G2 X-1.588 Y-237.154 Z-1.117 I-0.822 J3.067 K0.000 F60.00
G1 X-204.588 Y-119.952 Z-1.117 F60.00
G2 X-206.175 Y-117.202 Z-1.117 I1.587 J2.750 K0.000 F60.00
G1 X-206.175 Y117.202 Z-1.117 F60.00
G2 X-204.587 Y119.952 Z-1.117 I3.175 J-0.000 K0.000 F60.00
G1 X-1.588 Y237.154 Z-1.117 F60.00
G2 X1.587 Y237.154 Z-1.117 I1.588 J-2.750 K0.000 F60.00
G1 X154.844 Y148.671 Z-1.117 F60.00
G1 X154.844 Y148.671 Z-3.233 F60.00
G1 X204.587 Y119.952 Z-3.233 F60.00
G2 X206.175 Y117.202 Z-3.233 I-1.587 J-2.750 K0.000 F60.00
G1 X206.175 Y-117.202 Z-3.233 F60.00
G2 X204.587 Y-119.952 Z-3.233 I-3.175 J0.000 K0.000 F60.00
G1 X1.587 Y-237.154 Z-3.233 F60.00
G2 X0.822 Y-237.471 Z-3.233 I-1.588 J2.750 K0.000 F60.00
G2 X-1.588 Y-237.154 Z-3.233 I-0.822 J3.067 K0.000 F60.00
G1 X-204.588 Y-119.952 Z-3.233 F60.00
G2 X-206.175 Y-117.202 Z-3.233 I1.587 J2.750 K0.000 F60.00
G1 X-206.175 Y117.202 Z-3.233 F60.00
G2 X-204.587 Y119.952 Z-3.233 I3.175 J-0.000 K0.000 F60.00
G1 X-1.588 Y237.154 Z-3.233 F60.00
G2 X1.587 Y237.154 Z-3.233 I1.588 J-2.750 K0.000 F60.00
G1 X154.844 Y148.671 Z-3.233 F60.00
G1 X154.844 Y148.671 Z-5.350 F60.00
G1 X204.587 Y119.952 Z-5.350 F60.00
G2 X206.175 Y117.202 Z-5.350 I-1.587 J-2.750 K0.000 F60.00
G1 X206.175 Y-117.202 Z-5.350 F60.00
G2 X204.587 Y-119.952 Z-5.350 I-3.175 J0.000 K0.000 F60.00
G1 X1.587 Y-237.154 Z-5.350 F60.00
G2 X0.822 Y-237.471 Z-5.350 I-1.588 J2.750 K0.000 F60.00
G2 X-1.588 Y-237.154 Z-5.350 I-0.822 J3.067 K0.000 F60.00
G1 X-204.588 Y-119.952 Z-5.350 F60.00
G2 X-206.175 Y-117.202 Z-5.350 I1.587 J2.750 K0.000 F60.00
G1 X-206.175 Y117.202 Z-5.350 F60.00
G2 X-204.587 Y119.952 Z-5.350 I3.175 J-0.000 K0.000 F60.00
G1 X-1.588 Y237.154 Z-5.350 F60.00
G2 X1.587 Y237.154 Z-5.350 I1.588 J-2.750 K0.000 F60.00
G1 X154.844 Y148.671 Z-5.350 F60.00
G1 X154.844 Y148.671 Z-7.467 F60.00
G1 X204.587 Y119.952 Z-7.467 F60.00
G2 X206.175 Y117.202 Z-7.467 I-1.587 J-2.750 K0.000 F60.00
G1 X206.175 Y9.439 Z-7.467 F60.00
G1 X206.175 Y8.272 Z-6.300 F60.00
G1 X206.175 Y-8.272 Z-6.300 F60.00
G1 X206.175 Y-9.439 Z-7.467 F60.00
G1 X206.175 Y-117.202 Z-7.467 F60.00
G2 X204.587 Y-119.952 Z-7.467 I-3.175 J0.000 K0.000 F60.00
G1 X1.587 Y-237.154 Z-7.467 F60.00
G2 X0.822 Y-237.471 Z-7.467 I-1.588 J2.750 K0.000 F60.00
G2 X-1.588 Y-237.154 Z-7.467 I-0.822 J3.067 K0.000 F60.00
G1 X-94.913 Y-183.272 Z-7.467 F60.00
[... 95 lines cut ...]
G1 X-116.573 Y-170.767 Z-13.600 F60.00
G1 X-204.588 Y-119.952 Z-13.600 F60.00
G2 X-206.175 Y-117.202 Z-13.600 I1.587 J2.750 K0.000 F60.00
G1 X-206.175 Y-15.572 Z-13.600 F60.00
G1 X-206.175 Y-8.272 Z-6.300 F60.00
G1 X-206.175 Y8.272 Z-6.300 F60.00
G1 X-206.175 Y15.572 Z-13.600 F60.00
G1 X-206.175 Y117.202 Z-13.600 F60.00
G2 X-204.587 Y119.952 Z-13.600 I3.175 J-0.000 K0.000 F60.00
G1 X-1.588 Y237.154 Z-13.600 F60.00
G2 X1.587 Y237.154 Z-13.600 I1.588 J-2.750 K0.000 F60.00
G1 X64.730 Y200.699 Z-13.600 F60.00
G1 X71.052 Y197.049 Z-6.300 F60.00
G1 X85.379 Y188.777 Z-6.300 F60.00
G1 X91.701 Y185.127 Z-13.600 F60.00
G1 X154.844 Y148.671 Z-13.600 F60.00
G0 X154.844 Y148.671 Z6.000
(finish operation: Job)
(begin postamble)
M5
G17 G90
M2
`

It might very well be related to that issue.   I have not tried the cut and paste troubleshooting step.  I need to get a second SD card for the raspberry pi.  I have flashed it to use webcontrol at the moment.

rspeed, 2022-01-05:
Looks like Markdown formatting clobbered the newlines.

rspeed, 2022-01-05:
Oh, and I should note that I'm also getting the error on innocuous lines (including `G1`s), though I'm using Fusion 360 to generate it. The sled often does a small "wiggle" when it occurs, like the motors are drunk.

zaneclaes, 2022-01-05:
@rspeed you can use https://gist.github.com to paste & share files :)

> The sled often does a small "wiggle" when it occurs, like the motors are drunk.

hahah, interesting. Just to be sure, are you using the Maslow post-processor with Fusion360?

rspeed, 2022-01-05:
> @rspeed you can use https://gist.github.com to paste & share files :)

Oh, I meant the file that @buck2825 posted. I'll add my own gcode in a little bit when I have access to to the computer it's stored on.

> hahah, interesting. Just to be sure, are you using the Maslow post-processor with Fusion360?

Yup. I've also tried the GRBL post-processor with similar results.

rspeed, 2022-01-07:
Here's what the console reads:

> X-10.505 Y14.972 I-26.769 J-20.735
> ALARM:3 (Abort during cycle)
> position loaded at:
> -11.73
> 17.90

[The line where it stopped.](https://gist.github.com/rspeed/969fb3873de19f8e04f4fb38e3809421#file-cnc-room-grbl-nc-L35)

buck2825, 2022-01-09:
i copy and pasted the code and absolutely nothing happened.  
[image]

here is a screen picture.

rspeed, 2022-01-09:
> i copy and pasted the code and absolutely nothing happened.  

Huh. It went the wrong direction.

buck2825, 2022-01-09:
The image is what happens when I use the play button.  When I try to paste into the terminal the system does nothing.

greggbmt, 2022-02-05:
Was there a resolution for this?  I am having exact same issue

greggbmt, 2022-02-08:
Does anyone know if this has been resolved?  I am having exact same thing happen to me.

rspeed, 2022-02-11:
Nope.

buck2825, 2022-02-11:
@zaneclaes linked this bug report to the original code. https://github.com/cncjs/cncjs/issues/137  As that issue is closed I don't see anything happening soon.  

@zaneclaes I would open a bug report on cncjs but I don't have a clue how it integrates with Makerverse.

greggbmt, 2022-02-11:
I back-graded to webcontrol and my machine is working fine

rspeed, 2022-02-15:
I get the feeling like Makerverse is either sending the movement commands too quickly, or gets confused about which command is currently being executed. I assume the controller queues commands.

rspeed, 2022-02-19:
> I back-graded to webcontrol and my machine is working fine

Same here. I can run the same g code without any issues.