makermadecnc/makerverse #91: Makerverse crashing, dropping program file

RuckusGarage, 2021-12-20:
Seemingly randomly Makerverse will simply stop, the machine stops, the program is lost and the software is locked up. A PC restart is required and the entire program has to be run again from the beginning. This has been an ongoing issue for us for quite some time starting with the last build 1.1.3 and continuing into the current beta build we've been using 1.1.5. 

Our current SOP has staff restarting the computer and power cycling the Arduino in an effort to minimize crashes and the wasted time running programs multiple times. We've even gone so far as to break cuts into half hour to hour segments so as not to waste so much time when this happens. It goes without saying that this is frustrating and time consuming. 

Does the software log at all? Where would I find those?

Is there a known issue causing this problem and is there a solution? 

Best regards, RG

RuckusGarage, 2021-12-21:
Update; tried running the file again, crashed. Tried running a different file, crashed. Uninstalled and reverted back to 1.1.3 and tried again, crashed.

RuckusGarage, 2021-12-22:
Update: First cut of the day went fine. Second and third resulted in a new type of failure. The software crashed early on in the program but the program completed 100%. 

Reinstalled 1.1.5 and the fourth cut failed in a short time and in the original way.

RuckusGarage, 2021-12-23:
Tried an older file that's been cut successfully twice before today with the same result. After 2+ hours of cutting the machine just stops and the software dumped the program. At this point we've eliminated everything as a variable with the exception of the passage of time and the material being cut (different manufacture, same material). 

Are there known issues with the M2 Arduino Due failing? How about static discharge failures?

zaneclaes, 2022-01-04:
This sounds like a hardware issue. Specifically, this happens when the USB connection is interrupted for some reason. It could be a loose USB connection, bad cable, bad board, etc. Really, anything that would interrupt the connection stream with the machine.

> Does the software log at all? Where would I find those?

Software logs are printed to stdout.

closed 2022-01-04 by zaneclaes