makermadecnc/makerverse #93: Cannot load code

tgunr, 2021-12-22:
I have tried running MV and loading different file.nc on three different Macs always with the same result. After selecting a code file the "Loading" progress indicator just spins forever. I tried reinstalling from the web site, I tried running from docker all have same result. I tried building from source but get an error so I can't try that. 

The docker run only shows this:
`davec@iMacM1 ~/Dropbox/Sources/CNC/M2/makerverse
 % bin/launch                                                                                                                                                       (master)makerverse
Makerverse:latest settings: /Users/davec/.makerverse (project at: /Users/davec/Dropbox/Sources/CNC/M2/makerverse)
Docker v20.10.11 detected
Removing old versions (pruning docker)...
Total reclaimed space: 0B
Updating makerverse/core:latest...
latest: Pulling from makerverse/core
Digest: sha256:6eaf5ff53613a58f2130d9e5ea7dca608f024e1eb519a6c4a5f5c6182a09053e
Status: Image is up to date for makerverse/core:latest
docker.io/makerverse/core:latest
Killing prior instances...
Error response from daemon: Cannot kill container: makerverse: No such container: makerverse
Launching makerverse/core:latest...
2021-12-22T13:22:17.042Z - info init Loading configuration from "/home/node/.makerverse"
2021-12-22T13:22:17.109Z - info init Starting the server at http://172.17.0.2:8000`

I am at a loss and need this resolved, any ideas anyone?

skilescm, 2021-12-22:
Can you provide OS details on the Macs?

-Chris

skilescm, 2022-02-10:
Was this resolved?