makermadecnc/makerverse #95: Network Error - Can't Login

pauljacob95, 2022-02-06:
[image]

Can't login 
See picture, I've tried three devices across two internet connections in browser and in app. No results. Happy to trouble shoot things if helpful.

pauljacob95, 2022-02-06:
[image]

update, attempted updating to latest pre-release

pauljacob95, 2022-02-06:
[image]

kaibae19, 2022-02-06:
Me too.  This occurred yesterday afternoon Pacific time.  I lost the ability for two computers to authenticate over the course of the day.  Based on the CORS error, it suggests that the openwork.shop server was upgraded or changed as we are now missing expected headers for the openid authentication to work correctly.  The following error is also included in the web console:

`Failed to load https://openwork.shop/.well-known/openid-configuration: Redirect from 'https://openwork.shop/.well-known/openid-configuration' to 'https://www.openwork.shop/.well-known/openid-configuration' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://127.0.0.1:54720/' is therefore not allowed access.`

``

rspeed, 2022-02-15:
I've had some weird issues with logging in. Though it just brought me back to the login screen without any errors. The workaround I figured out was to go to the Open Workshop home page and clicking on the login button in the upper-right corner. Once I was logged in, returning to the Makerverse login worked correctly.

pauljacob95, 2022-02-15:
Cheers skilescm working for now, I have since been cutting on the machine without any interruptions. not sure when it came good.

Cheers for the feedback, kaibae and rspeed, 
Hopefully i wont have to try the login trick.

pauljacob95, 2022-05-01:
Works! Love you XoX

closed 2022-05-01 by pauljacob95

skilescm, 2022-05-01:
Can you check the console and see if it is a CORS error?