makermadecnc/makerverse #116: Network Error message at Login

HemiGuy, 2023-11-30:
Just installed Makerverse 1.1.33 and after startup I have the "Login (or Register)" button.  I click that and get a red text error message "Network Error"   Tried firewall settings, cleared cookies/cache etc nothing helped.

josephpaul96, 2023-12-02:
Facing the same issue.Were you able to figure it out?

josephpaul96, 2023-12-02:
[image]

HemiGuy, 2023-12-03:
I did... I put in a help ticket with Makermade and received a patched file... Directions and link below.

...................................................................................

You can still use the machine by bypassing the login and using Guest Mode in Makerverse.

To use Guest Mode you will need to find and replace the hidden (.makerverse) file in your User folder on your computer.

Attached to this email is an edited (.makerverse) file that you can replace it with.
The file location for Windows is: Local Disk (C:) > Users -> your user name -> .makerverse

Replace the (.makerverse) file that is already there with the (.makerverse) file that is attached to this email.
You will need to extract it from the zip file.

After you have replaced the file, open Makerverse and it should give you the option to use Guest Mode.
It will look similar to the screenshot link: https://drive.google.com/file/d/15JGEBhZOsePyRYSTXx_2vvjg-NHCTGze/view?usp=sharing
.............................................
Zip File (Patch) Link: https://drive.google.com/drive/folders/1507M1XevhCes9fSM_AKAfPRliLrMcXHH?usp=sharing